kaiaai/LDS #1: Problems with Delta2g

Davidm16, 2024-03-17:
I’m not sure how to connect the Delta2g lidar to an ESP32 or how to adapt the code for reading. It doesn’t work for me, help

kaiaai, 2024-03-17:
Delta-2g needs a little adapter board. Here is the schematic
https://github.com/makerspet/pcb under neato_delta_adapter/output Look for
the 5-pin  Delta-2A/B/G connector, connect TX to TX, GND to GND, MOT+ to
MOT+, MOT- to MOT-, 5V to 5V.

The adapter board connects to the ESP32 breakout board. Here is the
schematic https://github.com/makerspet/pcb/ under esp32_breakout/output .
Connect the 4-pin J3 from the Delta adapter board to the 6-pin LDS
connector on the ESP32 breakout board (same signal names to same signal
names).

As far as code goes, take the lds_basic_esp32 example included with the LDS
library. Change LDS_YDLIDAR_X4 to LDS_DELTA_2G.

Also, look at https://github.com/kaiaai/firmware under kaiaai-esp32 as
another example.

That said, I'm having a problem with Delta-2G. It runs and streams data OK,
bit the captured data shows some fixed pattern noise. I asked 3irobotics
for advice, but there is no answer. I know it can be fixed  because my
Delta-2 dev kit shows a clean picture, but I don't know yet how.

So for the time being, I don't recommend using any Delta-2 LiDARS, not
until this noise gets fixed. Overall, as a developer, I have a negative
impression from using Delta LiDARs - too many problems. Even their main
website https://www.3irobotics.com is down (but the http:// one still
works).

Davidm16, 2024-03-18:
Okay, I don’t think it’s worth the work of trying to run this lidar to detect just one obstacle that’s my goal. Do you know any cheap lidar that is easy to read with ESP32 or arduino? I just need to know the angle and distance of the sensor accurately

Davidm16, 2024-03-19:
Thanks, I’ll try it. With respect to Delta2g its's connected directly to an arduino mega serial I get a data packet of 11 length that does not change any. Should I not be able to read correctly if the engine and driver are properly powered?

kaiaai, 2024-03-20:
Delta-2G sends the 11 bytes length packet when the rotation speed too slow or too fast. Once you get the speed right, you should start getting long packets with distance data (and thus read it correctly).

closed 2024-03-20 by Davidm16