kaiaai/LDS #4: Support for RPLidar C1

sepastian, 2024-10-02:
Hello, thanks for this great library!

I own an RPLidar C1. The [documentation](https://files.waveshare.com/wiki/RPLIDAR-C1/SLAMTEC_rplidar_datasheet_C1_v1.0_en.pdf) states on page 7 that:

> The RPLIDAR C1 adopts traditional sampling protocols and the high-speed sampling
protocol for outputting the 5000 times per second laser range scan data. The
RPLIDAR C1 is compatible with all communication protocols used by previous
RPLIDAR S Series products and the traditional sampling protocol (standard) of the A
Series products, users can directly in place of previous RPLIDAR models and use it in
the original system. 

So, theoretically, LDS should be supporting the C1 already - correct?

I noticed that the C1 is using a baud rate of 460800, though. I changed that in https://github.com/kaiaai/LDS/blob/main/examples/lds_basic_esp32/lds_basic_esp32.ino, but cannot get the Lidar to return any data. The ESP32 doesn't seem to be able to connect to it.

Am I missing something (obvious)? 

Which steps would be involved in adding the RPLidare C1 to LDS?

Thanks!

sepastian, 2024-10-02:
Another question, since the hardware part is not my strength:

Do the Lidar and the ESP32 require a common ground connection?

I'm connecting the ESP32 to my PC via USB, to get serial output for debugging.

The Lidar is connected to a power supply.

Can that work?

kaiaai, 2024-10-03:
Hi sepastian,
  the LiDAR and the ESP32 grounds have to be connected together. Also
- LiDAR TX connects to ESP32 RX
- LiDAR RX connects to ESP32 TX
With these connections in place, I'd try again setting the baud rate and checking if you are receiving anything from the LiDAR. If you are not receiving, I'd double-check which ESP32 TX, RX pins you are using.
I wish I could give you a better advice, but I don't have a C1 to play with.

sepastian, 2024-10-05:
Thank you, I will give it a try as soon as I have time and let you know about the outcome.

sepastian, 2025-06-19:
That's great news, thanks! Will give it a try...

yudxp, 2025-09-04:
> I have it working with the C1. Changes were minimal.
> 
> 1. I copied the rplidar_a1_esp32.ino to rplidar_c1_esp32.ino
> 2. Copied LDS_RPLIDAR_A1.h and .cpp to LDS_RPLIDAR_C1.h and .cpp
> 3. I commented out lidar_motor_pin_callback() function and PWM related #defines
> 4. I set RX pin 16 and Tx pin 17 for my ESP32-S3 DevKitC 1.1 board.
> 5. added a new #define SLAMTEC_RPLIDAR_C1 in the ino file and lds_all_models.h file
> 6. Set baud to 460800 in LDS_RPLIDAR_C1.cpp
> 
> I left the enableMotor stuff as is for now and have not added anything new such as motor speed. The speed would be a new serial command, there is no motor PWM pin connector on the C1.
> 
> Here are the 4 changed files in this zip file.
> 
> [LDS_RPLIDAR_C1.zip](https://github.com/user-attachments/files/20816925/LDS_RPLIDAR_C1.zip)
> 
> The C1 requires 5V power. If the DevKit board is USB powered then the 5VDC pin will not have power. You need to supply 5V to the DevKit board externally or use a separate 5V supply for the C1 and tie the grounds together.

I try it, and it's working. Thank you