kaiaai/LDS #5: Add support for ESP32-C3 and LD19

minervasenior, 2024-11-03:
Hello, I’m using an ESP32C3 super mini and in theory it should be compatible but as this microprocessor is of RISC-V architecture and I am not able to receive by serial (UART RX pin 20). With a WEMOS ESP32 I can read the lidar LD14P although I am not able to read the C3.

Additionally, I see that the support for the LD19 lidar is not implemented yet

kaiaai, 2024-11-04:
Hi,
  I don't have ESP32-C3 or LD19 handy, so I cannot check this issue right away. That said, ESP32-C3 has only two UARTs, as opposed to ESP32 having 3 UARTS. So in the library sample sketch code you have to use UART1 for LiDAR (not UART2) and, for example, GPIO4 as RX1 and GPIO5 as TX.

  Try changing the library sample sketch code like this:
```
//ESP32-C3
//HardwareSerial LidarSerial(2);
HardwareSerial LidarSerial(1);
...
//LidarSerial.begin(baud_rate); // Use default GPIO TX 17, RX 16
LidarSerial.begin(baud_rate, SERIAL_8N1, 4, 5); // GPIO4 as RX1, GPIO5 as TX1
```
  Please let me know if this works or not.
Ilia

minervasenior, 2024-11-04:
It works! It was wrong serial. The library is compatible with this ESP. Thanks! I can't wait to get the LD19 working

closed 2024-11-04 by minervasenior