kaiaai/LDS #6: Library not found

ArkX23, 2024-11-06:
when i run the example ldrobot_ld14p_esp32  it shows a compiler error 
include <LDS_LDROBOT_LD14P.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: LDS_LDROBOT_LD14P.h: No such file or directory

kaiaai, 2024-11-07:
All looks fine on my end. Please check your library setup. Try opening the File->Preferences->Sketchbook location path, check it has library\LDS\ folders inside.

Here is my setup:

- Arduino IDE v1.8.19 for Windows
- Tools->Manage Libraries-> install LDS library v0.5.8
- File->Preferences->
  - set Additional Boards Manager URLs to `https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json`
  - sketchbook location is default `C:\Users\iYourUserName\OneDrive\Documents\Arduino`
 - Tools->Board->ESP32 Arduino->ESP32 Dev Module or ESP32 Arduino (Arduino)->Arduino Nano ESP32
 - open File->Examples->LDS->ldrobot_ld14p_esp32, click the Compile button. Compilation succeeds
```
Sketch uses 270477 bytes (20%) of program storage space. Maximum is 1310720 bytes.
Global variables use 21576 bytes (6%) of dynamic memory, leaving 306104 bytes for local variables. Maximum is 327680 bytes.
```

kaiaai, 2024-11-19:
Closing due to lack of follow-up. Reopen if needed.

closed 2024-11-19 by kaiaai