kaiaai/LDS #7: asking about my lidar

ErroujiOussama, 2024-11-19:
[image]
I have this lidar and I want to know how to use it with this library (wish version) the only information I know is that the lidar works with 115200 baud rate.
if you can help pls,
thank you.

ErroujiOussama, 2024-11-19:
is that working with esp8266 or arduino (mega&uno), cause I don't have the esp32?

kaiaai, 2024-11-19:
I have only tested it with ESP32.

ErroujiOussama, 2024-11-19:
ok, do you have a simple program to just test the lidar and read the data from it, also display it using an interface. send me all resources do you have aboutthis topic pls. 
thank you :).

kaiaai, 2024-11-19:
Please use the sample sketches in the 'examples' folder to run a Lidar and
print out its readings.

closed 2024-11-26 by ErroujiOussama