kaiaai/LDS #8: arduino ide ESP32

Frank-Kopek, 2024-11-19:
Dear kaiaai,
arduino ide uses a new version of the board management for the ESP32. The ledc ide now uses a different syntax. Would you please rewrite the examples with the new ledc syntax? Especially the "ledcSetup" is not working anymore. Thank You!

Regards

Frank

kaiaai, 2024-11-19:
Hi Frank,

  Espressif 3.0 compiler has changed some of its API and broke code that
uses the previous, 2.x API.

  I have tried code to 3.0 compiler, but there were some issues. I will try
again later when I have time.

  For now, please try building examples with Espressif 2.0.17 compiler.
Open the board manager, find ESP32 by Espressif and select the latest 2.x
version (2.0.17 I believe).

kaiaai, 2024-12-04:
I've added support for Espressif SDK 5.x (esp32 Arduino 3.x) in library release v0.6.0. Please try compiling. Reopen the issue if there is a problem.

closed 2024-12-04 by kaiaai