kaiaai/LDS #10: i use ydlidar x3 pro

sinjaeakang, 2024-12-11:
I add a Lidar sensor to the 'automatic_control example', and set with one channel and 360 points per second.

However, following test:
sensor.listen( lambda data: print( len(data) ) )
shows the sensor data has random length, arbitrarily from 0 to 480. Quite often it shows zero and some number <10. Only in some rare cases I get the most stable length as 359. I also tried no-rendering mode, and fixed the server fps to 20, but it doesn't help. Any idea to this?

kaiaai, 2024-12-11:
> channel and 360u points per second.

Hello,

what is the "automatic_control" example and sensor.listen(lambda..) code?
What is "server" fps?

  There is no example or code like that in the LDS Arduino library.

  It seems you are talking about some other library.

  I don't think I can help you with this.

kaiaai, 2025-01-14:
Stale issue, closing. Reopen if needed.

kaiaai, 2025-01-14:
Closing stale issue. Reopen if needed.

closed 2025-01-14 by kaiaai