kaiaai/LDS #11: YDLidar X4 scan_completed at 340 degree

Skara77, 2025-01-01:
Hi, with ydlidar x4 and esp32 scan_completed at 340 degree. How to modify the code to get 360 degree, means full scan with scan_completed.

kaiaai, 2025-01-01:
Hello,
  the code sends scan_completed when  X4 sends the scan-completed flag.
  If you are running a sample sketch, change the line 'if (i % 20 == 0) {' to 'if (i % 1 == 0) {'

Skara77, 2025-01-01:
Thanks for quick reply; I already did it before I shared the issue, So, it is not solution.  Any other idea?  Another comment; Do you plan to add pid motor control for ydlidar x4 library? I can not see it in LDS_YDLIDAR_X4.cpp

kaiaai, 2025-01-01:
Are you getting angles 340-360 *after* scan_completed?

Yes, I'd like to add X4 PID, but this is low priority. You are welcome to
contribute.

Skara77, 2025-01-01:
340 to 340, in reality; -20 degree to 340 degree. I am trying to add esp32 ros2 wifi micro-ros arduino. I can manage this issue in fact.  I already completed the code but in rviz2 scan message is vibrating. This is my main issue. When I will have time I will try to add pid control to ydlidar x4.

kaiaai, 2025-01-01:
Assuming you are handling scan_complete correctly when publishing /scan, is
it possible your X4 is damaged? Do you have another X4 to check?

X4 has a built-in PID. It keeps its speed rock steady - no
vibrating/wobbling.

Skara77, 2025-01-01:
I am checking scan_completed with your example; ydlidar_x4_esp32.ino. I don't have other lidar to check it. But works fine when connected to pc with usb adapter.  In my code, rviz2 scan image 4 - 5 degree shifting. This is my code;  

#include <micro_ros_arduino.h>
#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <sensor_msgs/msg/laser_scan.h>
#include <LDS_YDLIDAR_X4.h>

#if !defined(ESP32)
#error This example is only available for ESP32 Dev module
#endif

#define FRAME_ID "laser_frame"
#define TOPIC_NAME "scan"
#define NODE_NAME "uros_node"
#define BUF_LEN 1000 

#define LED_PIN 2
#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

const uint8_t LIDAR_EN_PIN = 19; // ESP32 Dev Kit LiDAR enable pin
const uint8_t LIDAR_PWM_PIN = 15; // LiDAR motor speed control using PWM
const uint8_t LIDAR_TX_PIN = 17;
const uint8_t LIDAR_RX_PIN = 16;

#define LIDAR_PWM_FREQ    10000
#define LIDAR_PWM_BITS    11
#define LIDAR_PWM_CHANNEL    2 // ESP32 PWM channel for LiDAR motor speed control

HardwareSerial LidarSerial(1);
LDS_YDLIDAR_X4 lidar;

rcl_publisher_t publisher;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;

sensor_msgs__msg__LaserScan laserScanMsg;
struct timespec tv = {0};
float tv_old = 0;
float tv_new = 0;

static float memory[BUF_LEN];
static float mem_intensity[BUF_LEN];
static float dist_offset = 0.15;

void error_loop(){
  Serial.println(F("error_loop()"));
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(500);
  }
}

void setup() {
  Serial.begin(921600);
  set_microros_wifi_transports((char*)"Wifi", (char*)"Password",
    (char*)"192.168.1.200", 8888);

  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  memset((void*)&laserScanMsg, 0, sizeof(laserScanMsg));

  laserScanMsg.header.frame_id.data = "laser_frame";
  laserScanMsg.header.frame_id.capacity = strlen(laserScanMsg.header.frame_id.data);
[... 133 lines cut ...]
      laserScanMsg.time_increment = (tv_new - tv_old)/i;
      laserScanMsg.scan_time = (tv_new - tv_old);
      tv_old = tv.tv_nsec/1.0e9;
      
      // LaserScan  
      laserScanMsg.angle_min = (min-360) * M_PI / 180; //0.0;               // start angle of the scan [rad]
      laserScanMsg.angle_max = max * M_PI / 180;//2 * M_PI;           // end angle of the scan [rad]
      laserScanMsg.angle_increment = (laserScanMsg.angle_max - laserScanMsg.angle_min) / i; // rad/meas
      
      
      rcl_ret_t rc = rcl_publish(&publisher, &laserScanMsg, NULL);
      if (rc == RCL_RET_OK)
      digitalWrite(LED_PIN, !digitalRead(LED_PIN));
      else
      Serial.println("rcl_publish() error");

       i = 0;
      }
     
  }

void loop() {

  lidar.loop();
}

kaiaai, 2025-01-01:
The code looks fine. That said, there should be no 4-5 degree or any
visible wobbling at all. My X4 has always been rock stable (as long as the
Lidar is NOT turning around its Z axis). So the only reason to add PID for
X4 is to set its rotation speed to something other than its default.

About X4 sending -20 to 340 degree data with respect to scan_complete, this
is typical for some low-cost Lidars. Sometimes 0 degree reading does not
even align with the physical Lidar's Y axis (you have to add a calibration
offset offset).

Skara77, 2025-01-01:
ok thanks, I think issue with cpu load. when no load, -13 to 347 degree, without any change. When cpu load increase with micro ros, it deviates -20 to 340, -15 to 345, -13 to 347. This angle deviation also creates deviation in rviz.

kaiaai, 2025-02-09:
Closing after a while. Reopen if needed.

closed 2025-02-09 by kaiaai